temcocontrols/T3000_Building_Automation_System #343: Calendar dates seems to be following International Date Line day (or perhaps an issue with Leap Year)

l-arnold, 2021-03-02:
Early in the day, Colorado,  Sync Controller Time with Local PC tends to either be off by 5 years, or be within about 1 hour, on subsequent logins and Control Opens.  Sync eventually matches the time, but the date stays off even though it had been set.

I note this when I open the controller from a separate T3000 computer.
Controllers do not have Internet access outside the local network so cannot test NTP time setting (I have tried in the past)

I note that "late in the day" new "Sync Controller to Local PC" tends to skew the date by one day.  This is with all being set to Mountain Standard Time USA.

These two screenshots show Time being set on one T3000, being followed by the same error, after time was set, on a second T3000 machine.
[image]

[image]

[image]

I would recomend having some information in the control panel about UTC time.  That is absolute time.

Fance, 2021-03-02:
The problem of a day and an hour time difference has been fixed in Rev54.4 by the firmware.Your firmware version is 54.3, please update your firmware。（When you update your firmware, make sure you are using the latest ISP Tool）
Thanks！
[image]

closed 2021-03-02 by Fance

l-arnold, 2021-03-02:
Working on this.  T3-LB continued with the issue after update of firmware.  Now doing T3-OEM.  What does the statement  "Change the priority of UART1 to avoid dead" mean.  I assume this refers to the RS485 port.  Explain if you would.

It seems I was able to update over Wifi (T3-OEM) but then the unit becomes inaccessible in T3000 and seems to "go to sleep".  Actually I see it has a "configure wifi" alert note.  Just restarted it now.  Seems to have wifi back but if this was inaccessible it would be complicated to get to the switch for restart.