temcocontrols/T3000_Building_Automation_System #350: problème de programmation

hamzalouh, 2021-03-26:
Hello,

I am in the programming part and I have some trouble with the basic programming.
I cannot control my valve which is the PID1 with respect to the heating curve which is y = -2.x + 70 and outside temperature.

Fance, 2021-03-29:
You can refer to the software instructions for the usage of some keywords
https://temcocontrols.com/ftp/software/10SoftwareManual.zip 

You can describe your problem in detail in this forum (software version screenshot), we will reply to you in time.
https://forums.temcocontrols.com/t/programming-for-bacnet-and-modbus-subnet-devices/156

MauriceDuteau, 2021-05-26:
> Hello,
> 
> I am in the programming part and I have some trouble with the basic programming.
> I cannot control my valve which is the PID1 with respect to the heating curve which is y = -2.x + 70 and outside temperature.

This would be a good question to post in the forum. The github repo is for development work. 
https://forums.temcocontrols.com/

closed 2021-05-26 by MauriceDuteau