temcocontrols/T3000_Building_Automation_System #424: Compiler Window Issues

lonerzzz, 2023-06-16:
In the latest release as of today, the T3000 compiler window has the following issues that were raised previously in the forums. I am raising here to improve the tracking for the following issues with the compiler window:

1) the compiler window cannot be resized meaning that on laptops, content gets cut off making development more difficult. The compiler window should be resizable.
2) only 1 compiler window can be open at a time which makes comparing code in different subprograms difficult
3) in the compiler window, when the settings colors are changed, the text goes to one color and is not updated with the new color selections
4) there is no compile button and no active compile indicator so sometimes it is unclear if the current content successfully compiles or not
5) the line numbers of the content are changed when the content is compiled meaning that organization of the code is harder to follow - line numbering serves as a way to put code into blocks for easier reading and the line numbering should be retained.
6) when the code is compiled, the window scroll position jumps making it confusing to follow where you as a programmer are working in the file
7) when the Save button is clicked, the name of the existing file is not shown by default. We have overwritten files accidently several times because we had to choose the file each time we saved.
8) there should be both a 'Save' and 'Save As' like standard windows programs so that a person can quickly save their work without having to pick a file. Only the 'Save As' or during a new file save would the user be asked for the file name,
9) if a program is too long, it should still compile the whole program to indicate all the errors and then also include an error about the program being too large.

Fance, 2023-07-07:
Item 1 ，Item 3 and Item 5   Done！

lonerzzz, 2023-07-15:
For item 5, the numbering is still changed when the code is loaded from the TStat10. I am not sure if this was changed or not, but the line numbers should be retained on the device also

lonerzzz, 2023-07-16:
For item 1, please allow the software to remember the adjusted size of the window for the duration of the program. Otherwise, it has to be adjusted every time it is opened which makes switching between programs more difficult.

MauriceDuteau, 2024-05-02:
Fandu will spend more time on this list.