temcocontrols/T3000_Building_Automation_System #425: Window Usage Issues

lonerzzz, 2023-06-16:
When using the T3000 software, the following windowing issues make the use of the software cumbersome in some respects:

1) The software does not maintain the windows that were open in their  position when the software is terminated so the windows must be open and resized whenever the software is terminated. The same is true of the column widths.
2) The software does not have a tiling option to neatly organize the open windows so that all are viewable
3) The PID window has empty space at the bottom that persists even when the window is shrunk down
4) The columns in different windows do not have a popup to describe the short form text in the column so it is not clear what the columns mean
5) The columns are not autoresized based on their content when the window is resized so empty columns are very wide and each column needs to be sized individually

MauriceDuteau, 2023-09-18:
Good comments, Fandu will work on the list.

MauriceDuteau, 2024-05-02:
Update: Apologies for not following up, some comments are addressed but not all. Fandu will spend some time on this asap.