temcocontrols/T3000_Building_Automation_System #428: Control Basic TBL function needs to handle variable lookup

lonerzzz, 2023-07-05:
The current Control Basic TBL functionality has been promised for some time but it is unclear how it would work. If it is only for static value lookup it is of limited use. The function should be able to take a variable memory location to represent the head of the table and an offset into that table in order to retrieve values. This would save considerable amounts of code where repeated 'if' structures are currently required to select the appropriate value in a table. The following is how it is now required to use variable based tables.

```
10 if value = 0 then lookupValue = var10
20 if value = 1 then lookupValue = var11
30 if value = 2 then lookupValue = var12
40 if value = 3 then lookupValue = var13
50 if value = 4 then lookupValue = var14
```

when it would really make sense to have the following instead:

```
10 lookupValue = tbl( value, var10 )
```

Between the the small program sizes and the lack of table functionality, far more code must be written to handle something relatively simple and common in programs

Fance, 2023-07-07:
I'll be working with our embedded engineers next week to figure out how to handle this list .Thanks！

lonerzzz, 2023-07-19:
Please retain this issue as a feature request as as the current implementation references fixed values which solves a different problem. Allowing table/array like access to variable values would be very useful functionality.

MauriceDuteau, 2024-05-02:
I will have our crew add some instructions on how to use the new TBL function, it is done but I cannot recall the details.