temcocontrols/T3000_Building_Automation_System #430: TStat10 Programming timeout is too long

lonerzzz, 2023-07-16:
When a TStat10 programming timeout occurs, the timeout takes about a minute to happen. I should take no more than 10 seconds to timeout.

MauriceDuteau, 2023-09-18:
Sorry to respond with a question but what sort of timeout are we talkikng about please. Bacnet, Modbus, RS485, IP, Wifi....

lonerzzz, 2023-09-18:
That is what I am referring to.

lonerzzz, 2023-10-09:
This issue continues to waste a lot of our time during development.

MauriceDuteau, 2023-10-09:
We checked and rechecked, the software waits three seconds before declaring a timeout. If you are seeing longer than that then perhaps the communications health is not good. You can do a wireshark capture and perhaps we can spot something from that.

lonerzzz, 2023-10-10:
This is what we get on the program window:
[image]

This is the timeout message that we get:
[image]

This is the messaging in Wireshark:
[image]

Nothing stands out as a message with a timeout error.

Fance, 2023-10-10:
I'd be happy to help you with that. There are many reasons for this problem, such as wifi signal quality, firmware version that needs to be updated, software version that needs to be updated, ip address conflict, some parameter Settings of the device are wrong, and so on.
If possible, I'd like to remotely assist you in identifying the root cause of the problem. You can contact us at sales3 at
 temcocontrols.com and we can schedule a time for remote help you.

MauriceDuteau, 2023-10-10:
Fandu confirms again that T3000 waits three seconds and retries, It will do this three times and then put up the timeout message after  9 seconds total. 

Fandu: You could make things a little more responsive by showing the three retires at least. 
            Do make sure its both RS485 and Ethernet that have this similar timeout setting please.

lonerzzz, 2023-10-10:
> I'd be happy to help you with that. There are many reasons for this problem, such as wifi signal quality, firmware version that needs to be updated, software version that needs to be updated, ip [...]

Hi Fance, would it be possible to add or enable logging for the T3000 a to show what is happening. I could then send you the log file after several runs. This would be easier because without addition capture of details, I suspect that you would see exactly what I do - just a frozen editor screen followed by a timeout popup window.

MauriceDuteau, 2023-10-11:
You could show us the communications tab along the bottom status line of T3000, it shows the communications health. 

Also let's just confirm from looking at your log that you are using Wifi. T3000 can be finicky when there's several adapters on your PC, temporarily disable all VPNs, virtual adapters and the wired Adapter and give that a shot. 

If you have an RS485 dongle you could give it a try to connect over RS485.

lonerzzz, 2023-10-11:
The communication tab shows no increment in the error count when this situation occurs. The error count remains unchanged before and after the timeout. This was observed 6 times in the last little bit and every time no error increment occurred.

There are no virtual adapters or VPN connections. I tried limiting to the single ethernet connection and the issue remained unchanged. The computer is using ethernet to the WiFi router and the WiFi router connects to the TStat10.

MauriceDuteau, 2023-10-11:
OK, it starts to come back to me. The wifi modules in use on the original version Tstat10 are not good. Fandu will see what we can do to timeout earlier. Itll be best to program over RS485 to start, then make small edits only over wifi. Sorry about this. 

Also we have a new wifi module in the works, real soon now.

closed 2023-10-11 by MauriceDuteau

lonerzzz, 2023-10-11:
Based on what I am seeing with no error messaging, I am not sure it is possible to conclude that the issue is related to the networking. I would suggest that printing activity information to what is now just the compiler error window might be inciteful. It would certainly make for a more user friendly interface.

Something like the following would be useful user feedback regardless of the error cause:
2023-10-11 09:17:21 Compiling program
2023-10-11 09:17:23 Compiled size: 1671 bytes
2023-10-11 09:17:24 Connecting with unit
2023-10-11 09:17:25 Sending program
2023-10-11 09:17:27 Send complete.

Fance, 2023-10-12:
192.168.0.115 is my Tstat10 wifi device .
I have tested for many times, and there is no problem with Tstat10 wifi communication, which is very smooth in my place. The size I read is 1377 bytes. I suspect that your wifi communication quality is not very good, I still hope to be able to help remotely, or you can send the entire wireshark communication, capture the packet and upload it to me for analysis. Thank you very much!
[image]

Fance, 2023-10-12:
The entire reading process of program I tested took only 0.9 seconds. All you are encountering may be other problems, I hope to get more useful information to help you solve your confusion

lonerzzz, 2023-10-15:
I doubt it is the WiFi at this point because no messaging is ever in progress in Wireshark when the windows hangs and the router is now right beside the thermostat. Is it possible to add more detail to the compile output window as suggested in my last message? It would be useful in general to know more about what the software is doing and would help to specifically address this issue.

MauriceDuteau, 2023-10-16:
It’s the wifi module on board the device, this module is known to be a poor performer. We will have a new version in the next iteration of our hardware coming out in a month or so. Could you give it a try over RS485, it will be fast and solid. Use the wifi connection once commissioning is done, for intermittent tuning adjustments. 

 

Fandu can add some debug messages as you described, keep an eye out for that in a release coming soon.