temcocontrols/T3000_Building_Automation_System #433: Error while debugging

VPKillerQueen, 2023-10-02:
Hi, I'm trying to debug the application for educational purposes, in the process, after I have connected to Modbus slave, I fail to start the control interface when double click into the device.
To be more precise, the error I have is "Exception thrown: read access violation. This was nullptr". "This" prefers to m_pCtrlSite in the winocc.cpp file. What should I do?
Thank you for your help.

VPKillerQueen, 2023-10-04:
**Exception thrown at 0x01017F35 in T3000.exe: 0xC0000005: Access violation reading location 0x00000074.
Unhandled exception thrown: read access violation.
**this** was nullptr.**
This is the error to be specific. Pls help

MauriceDuteau, 2024-05-02:
Sorry, we havent been tracking the issues feature here. Will get our team to spend time on it more regularly and check into this if you are still seeing the problem.

closed 2024-05-02 by MauriceDuteau