temcocontrols/T3000_Building_Automation_System #434: Feature requests based on Commissioning with the T3000 software

lonerzzz, 2023-10-15:
After having used the T3000 software to commission a thermostat on site in the last few days, we would like to suggest the following enhancements to make the software more amenable to use in such situations. What we found is that we were taking longer than needed because we could not trust the information presented at a point in time because we did not know if it was up to date. We know we had WiFi in the building that wasn't great but could do nothing to improve it so it is important that the T3000 software can better handle these situations.

1) each window should only request updates for the data currently visible in the window - for example, only 10 inputs are needed for a thermostat, but all registers are pulled which is inefficient and slows all the updates

2) there should be a time shown in the T3000 software and on each window (Inputs, Outputs, PIDS, etc), the last refresh time should be shown so that when calibrating an input for example, you can know that changes have been applied

3) there should be a refresh button in each window header to prompt refresh of only that window when the data in one window is more important at the moment for the person interacting with the system.

4) the window interactivity should not be compromised by a slow connection,

5) resizing the window and scrolling the window contents should should not cause a jump or hang based on data updates. It seems as if the double buffering of the window contents is not providing smooth interactivity or the window updates are in the same thread of execution as the network access code

MauriceDuteau, 2023-10-22:
Great feedback from the site, I like it. Fandu will work on this right pronto. Keep the feedback coming, anything to make things go faster on site is top priority.

Fance, 2023-10-24:
1. The refresh value is the data read by the block read command, and only reading certain areas does not reduce the number of reads or make them more efficient. These are the conclusions we came to after detailed testing.
2.Our UI is now so bloated that it's unlikely we're adding a new column to show when the data was last updated. I want to think about a more appropriate scheme to show when the data was last updated.
3.Click the input menu icon or output menu icon on the menu to refresh immediately, most people may not know this, I think I need to add a refresh button menu icon to let customers know how to operate the refresh more intuitively.
4. I don't really understand how interactivity is affected, but giving me more information allows me to identify and improve the problem.
5. Thanks！

lonerzzz, 2023-10-24:
For 1, the response does not hold logically based only on the limited detail provided in the response. If only the first 10 inputs or first 30 variables are read for example, then only that data is transferred. If the T3000 code is reading larger blocks including more variables, then the solution is to enable reading of smaller blocks with a defined range. The numbers of reads may not change but the amount of data transferred would be reduced. Similarly, if the TStat10 is bundling a number of values to send back, then it should bundle a configurable number of values to send back.

For 2, I am definitely not asking for a new column. I would suggest the time be on the window header itself and be updated once everything visible in the window was updated. The current time could be in the footer for reference.

For 3, clicking on the input or output buttons is not an intuitive way to refresh and it has side effects with regard to window placement. It would be better to have the button on the window header. In any case, there should be some feedback on the window that refresh is occurring.

For 4, I would suggest trying to use the T3000 software with the TStat10 with the router placed a long way away. You will see all sorts of issues with the UI being stuck for periods of time before becoming responsive again. If there is a place to drop screen captures, I can provide videos to show some of the issues.

lonerzzz, 2023-11-02:
To add to the suggestions, we would suggest two modes for the T3000 interactions and windows. One mode is a programming/setup mode where all the different registers can be changed. The second is an observation mode where everything except the values is not modifiable or updated after initial retrieval. This would limit the amount of data that needed be retrieved allowing for more rapid updates.

lonerzzz, 2024-04-16:
These issues continue to cause numerous problems on site making for painful interactions and adding delays to interactions with the thermostat. Another issue is that double clicking on the scroll bar causes the window to resize and is a real pain when working on a laptop screen. Such a double click should only resize the screen when the window header is clicked on. That is normal windows behaviour.